adafruit/seesaw #2: Building on Unix?

PTS93, 2017-10-17:
I was wondering if there are any dependencies that make the code tied to Atmel Studio (or at least make it harder to use something else) or if compiling with arm-gcc on say Linux should work just as well?

deanm1278, 2017-10-17:
@PTS93 we use Atmel studio for stuff like debugging, build script management, etc.
It will compile just fine with arm-gcc (that's what Atmel studio uses behind the scenes) but currently you will need to create your own makefile.

closed 2017-10-17 by deanm1278

PTS93, 2017-10-17:
Thanks for the info and quick reply!

deanm1278, 2018-05-30:
just saw this issue again. @PTS93 the build is makefile based now so you should be able to build on any system as long as arm-none-eabi compiler tools are in your PATH

PTS93, 2018-05-30:
oh thats cool! I got by with Atmel Studio so far but as I primarily use Linux the make file is a great help, thanks!