adafruit/seesaw #3: I2C Bootloader for firmware updates

jgilbert20, 2017-11-08:
@deanm1278, @tannewt really want to say I love the work on seesaw and have a technical proposal for improving the offering. I think the world needs a solid I2C-based bootloader that can be easily triggered by a user to load new code onto a SAMD11/09 device.

I'd be interested in co-developing this project with @adafruit. I've been working on my own [ecosystem](https://github.com/jgilbert20/Makernet) of I2C-based plug-and-play modules using ATSAMD11 (nearly identical to the ATSAMD09) ([see github here](https://github.com/jgilbert20/Makernet), and [Hackaday writeup here](https://hackaday.com/2017/05/27/hackaday-prize-entry-makernet/) ). 

Here is smattering of the devices and firmware I've built:

[image]. 

The first of these products is now on [sale on Tindie](https://www.tindie.com/products/jgilbert/makernet-knob/). 

For both these use cases (Adafruit/Seesaw and Makernet), we will have a lot of less-experienced Makers who will use these products and will not have the infrastructure such as jigs and JTAG programers to upgrade the firmware. The more systems Seesaw is incorporated, the more urges you'll have to improve but then you'll be stuck working out backwards compatibility. For Makernet, I also have the same issues, with the added complication that there is different firmware for each peripheral type and a more complex networking stack. For instance, future Makernet versions will allow device communication through a I2C/packet-radio bridge. 

I'd propose that Adafruit and Makernet join forces to build a bootloader and infrastructure for software upgrading ATSAMD11/09. I'd certainly be willing to contribute money towards hiring a programer to do this work and split the costs with Adafruit. Erin Tompson at Modulo actually has the basic framework already built as a starting point. 

Here is the specification I've been shopping around on Freelancer and other places for the bootloader:

### Bootloader Product Brief
Write a bootloader for the Atmel SAMD21 and Atmel SAMD11 that allows new firmware programs to be loaded over the I2C bus. This firmware will support a set of Maker/hobbyist products that I am developing which use the I2C bus for communications. (Example of a I2C-based product is here: [Makernet Knob from jgilbert on Tindie](https://www.tindie.com/products/jgilbert/makernet-knob/)) Since my end users will likely NOT have access to (or desire to use) a JTAG programmer, I’d like to provide them a simpler way to transfer new code for upgrades.  I’d like to enable a user to run a program on their local PC that streams a .HEX file to a Arduino-compatible board (like a Feather M0) and then that board would connect to the target device and transfer the program code.

One major part of this project is to write the code of the firmware, the transfer utility and the programmer firmware. Bootloader source code should be written in C or C++, compilable using the GCC toolchain, with comments in English. Compiled code must be smaller than 2KB, leaving 14KB for program code.

The second major part is to ensure this bootloader and toolset is integrated into a logical way into the existing open source tools. As part of this project, I’d like the freelancer to fork the MattairTech SAMD11 github repository and add in the option for the I2C bootloader as a project configuration option. It is important to be able to generate full firmware images (bootloader + firmware) from within the Arduino IDE so the process for doing that needs to be documented. This is to support side-loading applications.

Proposed end-user acceptance test cases:

1. User is writing new firmware for a board, and wants to generate a .HEX file that contains both the I2C bootloader and the new firmware. This bootloader+firmware .HEX would be flashed onto a device using a traditional JTAG programer, leaving the I2C bootloader so that future firmware updates can be made over I2C.
2. User has generated new firmware in the Arduino IDE and wants to push it to a device where the I2C bootloader has already been installed. In this case, there must be a way of preparing the HEX file with just the firmware (not the bootloader) and passing it to a script that transfers it to an I2C master. This second use case may require a custom linker script.

We'd want some new firmware loading sketchs, compilable in the Arduino 1.8 IDE and that can run on a ATSAMD21 board, that demonstrates sending new firmware from a PC to a target device using an existing SAMD21 board as a programmer. This could be a python script, or anything else as long as it can be clearly documented and runs on OSX and Windows. I proposed we'd have a single command line option can feed a new firmware image to an I2C Bus Master, which in turns spools the image to the I2C SAMD11/21 slave, and reports back to the command line if the bootloader exited cleanly. The bootloader should be written not to rely on any external clock or crystal and should use the watchdog timer for safety (see http://www.atmel.com/Images/Atmel-42205-SAM-D20-I2C-Slave-Bootloader_AP-Note_AT03304.pdf) Note that there is already a semi-working example of code that does this but it has extra stuff that needs to be removed. [GitHub - modulolabs/bootloader-samd: Bootloader for SAMD based modulos.](https://github.com/modulolabs/bootloader-samd) and [GitHub - modulolabs/bootloader-utils: Utilities for the modulo bootloader](https://github.com/modulolabs/bootloader-utils).

Code would ideally be written using direct register CMSIS access (would want simple documentation on each call), or if necessary, using the Atmel ASF framework (less preferred but acceptable). A single Makefile and source tree should be included that produces a new booloader image from a clean working directory. 

Note: Bossac support for the bootloader is not required. Program should verify the checksum of the loaded firmware block and provide an indication back to the loading code if the new firmware was successful.

pyr0ball, 2019-02-27:
Just popping in to see if there's been any progress on this?

I've got a project of my own where I'd like to utilize an i2c bootloader with an ATSAMD21 family MCU, and would be happy to assist with development

ladyada, 2019-02-27:
nope, no work that we know of - its all yours! :)