adafruit/seesaw #4: setPWMFreq() unresponsive.

wa1tnr, 2018-01-09:
Looking to run it at 50 Hz for a servo motor.  Seems to
want to stay on ~490 Hz or so (judging by ear with piezo
transducer connected to either Pin 5 or Pin 6.

No problem with fade.ino and the buzzer seems to annunciate
different pulse width modulations.  Just the base (carrier)
freq is fixed.

Recently shipped from Adafruit.

deanm1278, 2018-01-09:
hi @wa1tnr, PWM on the seesaw was updated to 16 bit and the ability to set the carrier frequency was added after the SAMD09 breakout was made. Please try your code again after updating to this firmware:

[seesaw_new_firmware.zip](https://github.com/adafruit/seesaw/files/1616082/seesaw_new_firmware.zip)

Sorry for the inconvenience!

wa1tnr, 2018-01-09:
Thank you, **deanm1278** -- I appreciate the fast response!  No worries -- Adafruit moves at lightspeed.

I will close this issue, though I cannot test the firmware (or don't know how, at this point) I'm sure it's right.  Fascinated to find out what it was (will look!)

Thanks again.  73 de wa1tnr

closed 2018-01-09 by wa1tnr

deanm1278, 2018-01-10:
@wa1tnr it can be programmed using the swclk, swdio, and reset pins. The easiest way to program is with a j-link or other SWD programmer, but if you don't have that we have a library that lets you use a cortex M0 arduino like a feather M0 or metro M0 to program another cortex chip (samd09 in your case).

https://learn.adafruit.com/programming-an-m0-using-an-arduino/overview

wa1tnr, 2018-01-13:
@deanm1278 - was able to follow your programming guide, no problem (used an Adalogger and an SD card, and the firware linked in this Issue).  Seesaw  Breakout now definitely responsive to setting of PWM frequency (as annunciated by a piezo buzzer, that revealed an obvious by-ear change in frequency -- and as demonstrated using a servo motor).
Thank you! -wa1tnr