adafruit/seesaw #6: Crickit - add red NeoPixel blink when power supply is > 5.5V or < 3.5V

ladyada, 2018-05-29:
seesaw runs even when the power supply is too high/too low because theres a 16V regulator, but it should blink the neopixel red if it detects it. maybe have it check once a second? there's a voltage divider on PB03 (VIN_SENSE)

pelikhan, 2018-05-29:
(green and red is brown for color blind users) Maybe it should blink SOS :)

ladyada, 2018-05-29:
hmm, @pelikhan good point -  what colors would it be for blinking SOS and what should it do when power is good?

pelikhan, 2018-05-29:
The current design is good. Solid/green: good, Blink/red: no good.

PTS93, 2018-05-29:
Sorry if I just swoop in here but I find that a great feature.
Personally, for bad power, I would go for very fast blinking (~100-200ms) then wait in off state for 800-1000ms and then repeat. This would signalise distress well imo and can't really be mistaken for some typical blinky example code as its a rather distressed signal that isn't uniformly blinking.

For good power I would keep it at steady green, steady LED usually means "all good". It would also be rather annoying to have the board consistently blink a pattern while having it sit next to you when developing even though everything is cool.

ladyada, 2018-05-29:
ok thanks everyone! we'll close when we've deployed

deanm1278, 2018-05-30:
ok this has been added! https://github.com/adafruit/seesaw/commit/35e168908343212d368241a43c3be3a3bb2e4bc7

The commit is a little longer just because I have added a general "system is conflicted" state. Everything works fine in conflicted state for now but the LED will flash red to alert the user. If everything is fine, the led will be solid green.
In this case the system will enter conflicted state (flashing led) when the voltage goes below 3.7V or above 5.7V

I like the suggestion from @PTS93 and the LED will flash fast red and then turn off for a few seconds in a specific pattern to signal distress.

these are the board-specific configuration settings for the power sense LED option. https://github.com/adafruit/seesaw/blob/master/boards/crickit/board_config.h#L174

Note that for the the LED only changes when the system changes state (enters or exits conflicted), so the user can overwrite the solid green neopixel with their own data using the regular neopixel API.

closed 2018-05-30 by deanm1278