adafruit/seesaw #7: Build fails - missing build_date.h

jerryneedell, 2018-06-02:
Trying to build for BOARD=crickit for BAORD=samd09Breakout fails due to missing include file.

```
In file included from lib/qp/extras/fw_log.cpp:36:0:
./include/SeesawConfig.h:7:10: fatal error: build_date.h: No such file or directory
 #include "build_date.h"
          ^~~~~~~~~~~~~~
compilation terminated.
Makefile:148: recipe for target 'build/crickit/lib/qp/extras/fw_log.o' failed
make: *** [build/crickit/lib/qp/extras/fw_log.o] Error 1
```

deanm1278, 2018-06-02:
hey @jerryneedell, that file is auto-generated for each build. Make sure you have python in your path and can run
```python scripts/datecode.py .```
from the repo root directory

also if you're looking for the most recent crickit build the uf2 can be found here: https://github.com/adafruit/seesaw/releases/tag/1.1.2

jerryneedell, 2018-06-02:
@deanm1278 Thanks - I was jsut testing to see if I could build it.
I tried:

python scripts/datecode.py .

and it runs ok but the build still fails.

When is the python script run to create build_date.h?

deanm1278, 2018-06-02:
the script is run here: https://github.com/adafruit/seesaw/blob/master/Makefile#L120
when you run the python script does the build_date.h file get created in the repo root?

jerryneedell, 2018-06-02:
No - nothing is created.

deanm1278, 2018-06-02:
maybe a permissions issue and it can't create the file? Try running as root?

jerryneedell, 2018-06-02:
it works if I make this change:
```
-f = open(sys.argv[1] + '\\build_date.h', 'w')
+f = open(sys.argv[1] + '/build_date.h', 'w')
 
```

jerryneedell, 2018-06-02:
I am building on Linux (also tried MacOS)  is the script assuming Windows? (uses "\")

deanm1278, 2018-06-02:
oh yea windows paths. We can probably remove those slashes entirely... I'll try on monday
does the rest of the build work once you fix that file?

jerryneedell, 2018-06-02:
Yes - it seems to work fine with that change - No errors with 
make all-boards
Thanks - no rush on this. I just wanted to make sure I understood what was going on.

deanm1278, 2018-06-13:
fixed!
https://github.com/adafruit/seesaw/commit/46764c7001897b8a43d3bbc0a83a40ecde81828b

closed 2018-06-13 by deanm1278