adafruit/seesaw #8: Servo motor doesn't go to 0

danowall, 2018-06-05:
In MakeCode, servo motor doesn't go all the way to 0 degrees when "crickit set servo angle to 0" is used. 

Using "crickit set servo pulse to 500" does send servo to 0 degrees, however.

ladyada, 2018-06-05:
this isn't a bug, this is just how servos work :)
https://learn.adafruit.com/adafruit-crickit-creative-robotic-interactive-construction-kit/makecode-servos#precise-pulses

https://learn.adafruit.com/adafruit-crickit-creative-robotic-interactive-construction-kit/circuitpython-servos#min-slash-max-pulse-control

danowall, 2018-06-05:
got it! thank you for these links, this is helpful.

closed 2018-06-05 by danowall

deanm1278, 2018-06-05:
hey @ladyada did we test that setAngle gives the same results as the corresponding function in cirpy or arduino?

ladyada, 2018-06-05:
we didn't but we can. in cirpy you can adjust the min/max pulse in the servo library and i did that a lot.
we also default to 1-2ms in circuitpy
https://github.com/adafruit/Adafruit_CircuitPython_Motor/blob/master/adafruit_motor/servo.py#L43