adafruit/seesaw #11: Help getting uf2 file

kstisser, 2018-09-20:
I am looking to update the firmware, and was following this guide which led me to your page: https://learn.adafruit.com/adafruit-crickit-creative-robotic-interactive-construction-kit/update-your-crickit. However, they say drag the uf2 file onto the board. I did not see a uf2 file in any of the releases and it was not evident how to build one. Can you please help clarify how to get the uf2 file?

ladyada, 2018-09-20:
oops its here! https://github.com/adafruit/seesaw/releases/tag/1.1.4

ladyada, 2018-09-20:
@deanm1278 do we need a release for neotrellis?

deanm1278, 2018-09-20:
I'd like to have releases with the production firmware for each board if
possible.

ladyada, 2018-09-21:
problem is we link to 'latest' in the crickit guide, but latest is neotrellis only :/

kstisser, 2018-09-23:
Thank you for the quick response. However, I tried to flash the new file, but am getting an error: 
E:\ is not accessible. The request could not be performed because of an I/O device error. 
I followed the tutorial, with only the USB to microUSB in the Seesaw only connection to the computer. The Green light is on, and the yellow light is pulsing, but when I try to do anything with the windows explorer with the E drive it freezes up on me for a long time. When I try to copy the file you directed me to I get the error I mentioned above. Do you what might be wrong?

ladyada, 2018-09-23:
if you're running some sort of antivirus sometimes that confuses it - try disabling it or trying another computer, just drag the UF2 only!

genus1, 2018-10-06:
I just performed the update with the UF2.  Everything went okay; however, the cricket now flashes red about 5 times, pauses then continues to repeat.  Is this normal or in the red flashing instead of solid green light something is wrong?

ladyada, 2018-10-06:
that means the DC power is bad :)

ladyada, 2018-10-07:
@deanm1278 this came up a few more times so i deleted the latest release

closed 2018-10-07 by ladyada