adafruit/seesaw #12: Would it be possible to have the seesaw borad in sleep mode

pierrot10, 2018-12-17:
Good day,
I am actually working with a Feather MO adalogger board and a Seesaw board.
When  remove the Seesaw board from the proejct, the project cunsome about 10mA less.

I can not determine how much cunsume the seesaw board, but would have a possiblity to have the Seesaw board sleeping and to wake it up as we can do with a Feather MO board and the ArduinoLowPower and the Adafruit_sleepdog libraries?

In other word, how can we have the Seesaw board in a low power cunsumption when the Feather MO board sleep?

pierrot10, 2018-12-18:
Dear Layada, 

Thank for your reply. Your answer raised the doubt :). May I propose you to add the Low Power functionality in the next SeeSaw version?
Is a complex task to add the Low Power mode? Is some code to add in the SeeSaw library or should it be modified in the Frimeware?

by the way, may I suggest again a new Learn section (if it's not not exist yet) about the Seesaw customization with a J-Link and SWD adapter?

Thank a lot for your support and I already wish a merry Christmas.

ladyada, 2018-12-19:
it would be great to add it but we don't have any plans to and the schedule's very full. :)
if you do end up adding that ability please send a pull request and we'll integrate it!