adafruit/seesaw #16: ENABLE_LOGGING - build failed.

wallarug, 2019-03-03:
Hi,

I am having an issue building SeeSaw with logging enabled.  When I attempt to `make BOARD=feather_m0` or any other board, the build fails.

**StackTrace**
```
In file included from source/AOTouch.cpp:35:0:
source/AOTouch.cpp: In static member function 'static QP::QState AOTouch::Root(AOTouch*, const QP::QEvt*)':
./include/event.h:41:38: error: 'Log' has not been declared
 #define LOG_EVENT(e_)            Log::Event(me->m_name, __FUNCTION__, GetEvtName(e_->sig), e_->sig);
                                      ^
source/AOTouch.cpp:120:13: note: in expansion of macro 'LOG_EVENT'
             LOG_EVENT(e);
```

StackTrace continues for all references to LOG_EVENT.  There is a problem with the namespaces but I have not been able to narrow it down.

Extract from **event.h**
```
#ifdef ENABLE_LOGGING
#define LOG_EVENT(e_)            Log::Event(me->m_name, __FUNCTION__, GetEvtName(e_->sig), e_->sig);
#else
#define LOG_EVENT(e_)
#endif
```

ladyada, 2019-03-03:
not sure - but it did work at some point! if you figure out the change needed to have logging work, please submit a PR :)

wallarug, 2019-03-05:
I solved this - completely my fault.

DEBUG=1 is required in the make command as per the documentation in README.md

```
make -j8 BOARD=crickitHAT DEBUG=1
```

There is no mention in the README.md if ENABLE_LOGGING flag actually does anything.  It must automatically get added.

As a side effect of enabling DEBUG, any of the adafruit_crickit libraries stop working.  They must require a fast I2C connection.  I will experiment with that further.

This issue is resolved.

closed 2019-03-05 by wallarug