adafruit/seesaw #18: Buffered/latched mode for buttons

deshipu, 2019-04-17:
In some cases we are not interested in the state of the pins *right now*, but instead we want to know if any of them changed state (even for a moment) since we last checked. This is especially useful when checking for button presses, as then we can have long delays between checks and still not miss any presses.

Would it be possible to add a mode to the pins in which they would "latch" their state, and keep it until either explicitly cleared with a separate command, or simply read? Pin with a pull-up would latch in the low state, and pins with a pull-down would latch in the high state.

This would let me write a dummy GamePad class for seesaw for games.

ladyada, 2019-04-17:
if you'd like to add it, please try! right now we do have an IRQ pin and reading all GPIO is a single I2C call

deshipu, 2019-04-17:
I can try, it will be a while though.

ladyada, 2019-04-17:
thanks - seesaw is currently in dev statis (works great, so not much for us to add) but we welcome any improvements