adafruit/seesaw #23: PulseIn Support

wallarug, 2019-05-06:
As mentioned on [Adafruit_CircuitPython_Seesaw GitHub](https://github.com/adafruit/Adafruit_CircuitPython_seesaw/issues/30), but more relevant here as it needs to be implemented in the firmware.

SeeSaw does not have a mechanism for reading pulses from the ADC pins (AFAIK).  This functionality is already built into CircuitPython, however, it doesn't understand how to communicate with a SeeSaw device for achieving this goal.

I want to be able to read the signal coming from my RC Controller with is sending a series of pulses, rather than analog input.

When attempting to do a pulseio read, it says "pulse io is not supported on this device or does not exist" (running on a Raspberry Pi).  There is nothing in the SeeSaw CircuitPython library either that is able to achieve this functionality.

My recommendation is that it be implemented directly into SeeSaw and just send back the values like it does for Encoders and ADC raw_values.

I'm not sure how much of the [existing C code for Pulseio.PulseIn](https://github.com/adafruit/circuitpython/tree/master/ports/atmel-samd/common-hal/pulseio) can be transferable to run within SeeSaw.  @tannewt do you have any ideas?

Happy to look into this.

- [ ] Check for existing support
- [ ] Check for portability between CircuitPython PulseIn and SeeSaw
- [ ] Attempt Implementation.

[Documentation for PulseIn](https://circuitpython.readthedocs.io/en/3.x/shared-bindings/pulseio/PulseIn.html)

ladyada, 2019-05-06:
hiya, there's nobody currently maintaining on or working on seesaw, if you want some more capabilities in seesaw you will have to implement them on your own - we have lots of code you can use, but we wont be able to dedicate any time soon to assisting you.

chatelao, 2021-01-14:
Same here: The signal I/O connectors would be perfect for reading RC-signal to add RC support to my robot.