adafruit/seesaw #29: Build failing with rom,ram overflow on Ubuntu 18.04

jhh, 2019-11-01:
I'm attempting to build on Ubuntu 18.04 and the build is failing. I'm using the stock compiler installed with `apt install gcc-arm-none-eabi`. Here's the specific error, any help would be appreciated.
```
❯  make BOARD=samd09Breakout
# ...
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: build/samd09Breakout/seesaw-samd09Breakout.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: build/samd09Breakout/seesaw-samd09Breakout.elf section `.stack' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 524 bytes
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 288 bytes
collect2: error: ld returned 1 exit status
Makefile:159: recipe for target 'build/samd09Breakout/seesaw-samd09Breakout.bin' failed
make: *** [build/samd09Breakout/seesaw-samd09Breakout.bin] Error 1

```
Compiler info:
```
❯ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

❯ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jhh, 2019-11-03:
I was able to successfully build on a Ubuntu 16.04 vm with the toolchain downloaded from the location listed in `.travis.yml`. Feel free to close this unless you would like to track this issue.

caternuson, 2019-11-03:
Try this on 18.04:
https://learn.adafruit.com/building-circuitpython/linux#install-build-tools-on-ubuntu-2-2

jhh, 2019-11-03:
Thanks @caternuson , that works as well.

closed 2019-11-03 by jhh