adafruit/seesaw #34: Is the SAMD10 tested with Neopixels?

john--, 2020-01-24:
I have a SAMD10 Xplained Mini board from atmel and was hoping to create a project using seesaw that incorporates touch and neopixels (touch is why I want SAMD10 over the SAMD09). 
I based my config off the soil sensor but enabled neopixel support. I can post it if that's helpful.
I connected to it over i2c with an arduino uno and tried specifying pin 25 (PA25) to light up my neopixels but haven't had any luck. 
Touch support and everything seems fine, but only the first neopixel lights up.
I could be doing something wrong, but I'm curious if anyone else has tested neopixel support on the SAMD10? 
The MCU seems similar enough to the SAMD09 that I would have figured the timings would be fine out of the box, but who knows...

ladyada, 2020-01-24:
no reason it would not - however will note we provide no tech support for DIY builds with seesaw, its not easy to bring up a new board and may require using a debugger :)

closed 2020-01-24 by ladyada

john--, 2020-01-24:
That was a fast response. That's fair, figured it was worth asking.
On an unrelated note you should totally sell sessaw dev boards with SAMD10s. Touch support is a wonderful thing.

john--, 2020-01-24:
Good news! I was able to get it to work by using pin 10 instead of 25.

In #17  @wallarug mentioned PA08, PA09, PA24, PA25 being problematic with TCC in SAMD10. I have no idea if that's applicable here, but I deliberately avoided these pins and the neopixel strip works now on PA10. As a software dev I should probably care why, but honestly I'm just happy it works. Thanks for being my sounding board @ladyada . And seriously, make a seesaw breakout board with a SAMD10 and I'll buy a bunch :)