adafruit/seesaw #40: Updating QP Framework

wallarug, 2020-04-04:
Hi there!

I have started to try and update the QP framework for SeeSaw as well but hit an issues that has stumped me.

**Console**
```
$ make BOARD=debug
Building debug
lib/qp/extras/fw_evt.cpp
In file included from lib/qp/extras/fw_evt.cpp:31:0:
lib/qp/extras/fw_evt.h: In constructor 'FW::Evt::Evt(QP::QSignal, uint16_t)':
lib/qp/extras/fw_evt.h:48:36: error: no matching function for call to 'QP::QEvt::QEvt(QP::QSignal&)'
         QP::QEvt(signal), m_seq(seq)
                                    ^
In file included from lib/qp/ports/arm-cm/qxk/gnu/qep_port.hpp:46:0,
                 from lib/qp/ports/arm-cm/qxk/gnu/qf_port.hpp:106,
                 from ./lib/qp/include/qpcpp.hpp:47,
                 from ./lib/qp/include/qpcpp.h:43,
                 from lib/qp/extras/fw_evt.cpp:30:
./lib/qp/include/qep.hpp:209:12: note: candidate: QP::QEvt::QEvt()
     struct QEvt {
            ^~~~
./lib/qp/include/qep.hpp:209:12: note:   candidate expects 0 arguments, 1 provided
./lib/qp/include/qep.hpp:209:12: note: candidate: constexpr QP::QEvt::QEvt(const QP::QEvt&)
./lib/qp/include/qep.hpp:209:12: note:   no known conversion for argument 1 from 'QP::QSignal {aka short unsigned int}' to 'const QP::QEvt&'
./lib/qp/include/qep.hpp:209:12: note: candidate: constexpr QP::QEvt::QEvt(QP::QEvt&&)
./lib/qp/include/qep.hpp:209:12: note:   no known conversion for argument 1 from 'QP::QSignal {aka short unsigned int}' to 'QP::QEvt&&'
Makefile:186: recipe for target 'build/debug/lib/qp/extras/fw_evt.o' failed
make: *** [build/debug/lib/qp/extras/fw_evt.o] Error 1

```

**fw_evt.h**  _Lines 40 to 55_
```
namespace FW {

class Evt : public QP::QEvt {
public:
    static void *operator new(size_t s);
    static void operator delete(void *evt);

    Evt(QP::QSignal signal, uint16_t seq = 0) :
        QP::QEvt(signal), m_seq(seq) 
	{}
    ~Evt() {}
    uint16_t GetSeq() const { return m_seq; }

protected:
    uint16_t m_seq;
};
```

I have not changed the `extras/` folder because that is included from a different [project?](https://github.com/galliumstudio/platform-stm32f767-nucleo) (which is not very well documented).   Purely just updating the framework and MakeFile produces this error.  It is on this [branch](https://github.com/wallarug/seesaw/tree/wallarug/qt6.8.0).

I don't understand it.

@deanm1278  Where did you originally source this fw_ code from?

deanm1278, 2020-04-04:
Adding `#define Q_EVT_CTOR` to this file may fix that compile error https://github.com/wallarug/seesaw/blob/wallarug/qt6.8.0/lib/qp/ports/arm-cm/qxk/gnu/qep_port.hpp

wallarug, 2020-04-05:
Thanks Dean!  Worked like a charm!

There was an error in `include/qep.hpp` on line 169 but that was easy to fix (extra bracket).  I'll report it over at the Quantum Systems repo.

Now to update all the AO*.cpp files in seesaw.

wallarug, 2020-04-05:
Hey,

I've hit a new issue now.   It seems to be the final hurdle.

[Makefile](https://github.com/wallarug/seesaw/blob/wallarug/qt6.8.0/Makefile)

**Output**
``` bash
In file included from lib/qp/ports/arm-cm/qxk/gnu/qf_port.hpp:106:0,
                 from lib/qp/ports/arm-cm/qxk/gnu/qxk_port.cpp:41:
lib/qp/ports/arm-cm/qxk/gnu/qep_port.hpp:48:10: fatal error: qep.hpp: No such file or directory
 #include "qep.hpp"  // QEP platform-independent public interface
          ^~~~~~~~~
compilation terminated.
Makefile:169: recipe for target 'build/debug/seesaw-debug.bin' failed
make: *** [build/debug/seesaw-debug.bin] Error 1
```

I have already updated the Makefile to the new paths for QP.   It doesn't seem to want to pick up the qep.hpp file for some reason.

Any assistance would be awesome!

deanm1278, 2020-04-06:
maybe something to do with it being compiled as an assembly file? try moving it from SSOURCES to COMMON_SRC to compile it along with the regular cpp files.

I will say though, you likely won't get any benefit from updating QP to the latest version. In fact if the kernel size gets larger you might even run out of room on the smaller chips. Is there a reason for updating?

wallarug, 2020-04-06:
Hey @deanm1278 ,

I tried your suggestion but didn't have any luck.  I moved `qxk_port.cpp` to the common src but it still fails.   

```bash
make: *** No rule to make target 'build/debug/lib/qp/ports/arm-cm/pxk/gnu/pxk_port.o', needed by 'build/debug/seesaw-debug.bin'.  Stop.
```

In response to your question:   I wanted to ensure that SeeSaw could be supported going forwards and I figured part of that is making sure everything included in it is up to date.  I have some plans to use SeeSaw going forward with some major projects and want to ensure that updating it (and the included libraries) is straight forward.    It may be true that the library is larger like you stated (which may cause problems).  I can test this once it is up and running.   I also note that some features are being stripped out soon, that would make SeeSaw completely un-update-able in the near future.

I appreciate your recent support and all the effort you have put into it over the years.  

SeeSaw is really quite powerful and understated in my opinion on what it can do.  I was surprised when it dropped off the top downloaded libraries last year on the Adafruit Weekly emails.

deanm1278, 2020-04-06:
hmm from the error you sent it looks like you wrote `pxk` and `pxk_port` instead of `qxk` and `qxk_port`
note you also may need to clean the project before running make.

wallarug, 2020-04-07:
Thanks Dean.  It compiles now 👍 .   

Apologies for not double checking that.

closed 2020-04-07 by wallarug