adafruit/seesaw #41: Seesaw UART messing with RPi serial0 UART.

jgoergen, 2020-10-03:
Cross posting (https://github.com/adafruit/Adafruit_CircuitPython_seesaw/issues/54)[this issue], just incase the other place I left it wasn't the correct repo.

I'm trying to read a sensor over standard 'minicom' serial0 UART on the Raspberry Pi, but the data only comes in when I hold down the reset button on the Crickit. If I plug a USB cable into the seesaw USB port and 'Putty into' the serial adapter, then connect to the Rasbperry Pi serial0 UART with screen, I can type data back and forth. So I know the Seesaw is, by default, connected directly to the Raspberry Pi UART pins.

Is there some way to disable this without physically cutting the lines on the Crickit board?
Thank you!

ladyada, 2020-10-03:
not really, you are best off cutting the TX trace

closed 2020-10-03 by ladyada

wallarug, 2020-10-04:
@jgoergen 

You can compile SeeSaw yourself without the Crickit using those pins for the UART.

jgoergen, 2020-10-04:
Alright, that's what I figured. 
Thanks @ladyada and @wallarug for getting back to me!