adafruit/seesaw #51: Encoder can lose counts

jepler, 2021-05-13:
With the Adafruit Rotary Trinkey and the suggested 24-detent-per-revolution encoder, run one of the sample programs which shows the accumulated encoder counts.  I used CircuitPython.

Carefully rotate the encoder about 1/2 to 3/4 of the way between detents, then return it to the initial position without actually passing over to the next detent.  Repeat this multiple times, always in the same direction (e.g., initially turning clockwise).  The encoder count will "drift away" from the initial count, even though the position is not changing.  Once you get the hang of it you'll be able to lose a count almost every time.

I suspect there's some kind of logic error in the quadrature decoder state machine, but its form was unfamiliar to me so I was not able to make sense of it.

ladyada, 2021-05-13:
do you mean the trinkey or the i2c breakout?

jepler, 2021-05-13:
Adafruit I2C QT Rotary Encoder with NeoPixel - STEMMA QT / Qwiic -- I updated the initial comment too

jerryneedell, 2021-05-22:
I have also reproduced this  with the I2C QT Rotary Encoder (PID 4991) on an QTPY  RP2040 with Circuitpython and a QTPY M0 with Arduino.