adafruit/seesaw #52: Encoder increments in the wrong direction CW decrements CCW increments

wifijt, 2021-05-18:
The current implementation of the SEESAW encoder increments in the opposite direction one might expect. Clock wise on most devices increases and Counter clockwise decrements. 

While it's possible to program around this - it would be better for most hobbyists if it incremented in CW and decremented CCW

evtimDev, 2022-07-26:
What encoder model are you using? From my experience, this depends on the particular encoder - some models have the waveform one way, and others have the waveform the other way - so the software will increment in either CW or CCW.  If you need to compensate/fix that you can simply swap the A & B pins - either in hardware or software.

ladyada, 2022-08-01:
its currently in the code as-is, so just swap pins please :)

closed 2022-08-01 by ladyada

leastbad, 2023-03-14:
> you can simply swap the A & B pins

@evtimDev unfortunately, this is so *simple* that there's no documentation on how to best accomplish this 😉 and unfortunately, [this is a persistent issue for lots of folks](https://forums.adafruit.com/viewtopic.php?t=179374).

Is there a way to redefine `CONFIG_ENCODER0_A_PIN` and `CONFIG_ENCODER0_B_PIN` without manually changing the library (which seems like it would break encoders that are currently working correctly)?

Are you talking about something like changing the `variant.h` for the board? This also seems heavy-handed, and it'd need to be tweaked any time there was an update.

It really does seem like exposing an option to configure directionality at the seesaw library level would be a reasonable approach, no?