adafruit/seesaw #53: I2C Documentation

BleuLlama, 2021-06-16:
Is there documentation of all of the I2C calls/spec that are supported here?  I looked through this and other seeseaw projects but couldn't find any.  and of course, if one of those other projects is the correct place to request I2C interface doucmentation, please let me know and i'll happily close this issue here.

I have one of the rotary + neopixels with seesaw on it, which I just got last week, and I want to just talk to it directly from my application without using or requiring installing library.  

For example, I can read the encoder by polling
"i2ctransfer -y 1 w2@0x36 0x11 0x40 r4"
or 
"i2ctransfer -y 1 w2@0x36 0x11 0x30 r4" 

ie, To I2C addres 0x36, write the two bytes:
0x11 - Encoder Main Address
0x30 - Encoder Read Value (or 0x40) for absolute position or delta from the last call
then read back the 4 byte value, which is the delta.position

but past that, i'm not sure how to set the gpio for the button or configure the neopixel, etc.  I tried reverse engineering the protocol for the version number request but couldn't get that to work for some reason.

ladyada, 2021-06-16:
did ya check out https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout/using-the-seesaw-platform ?

BleuLlama, 2021-06-16:
I hadn't found that page. Perfect! thank you!

closed 2021-06-16 by BleuLlama