adafruit/seesaw #54: two encoders at once?

troylball, 2022-02-01:


troylball, 2022-02-01:
Hi I am looking for some sample python code that will allow two encoders at the same time.  I have already solders the back to change one to a different i2c #,  any help would be appreciated

troylball, 2022-02-01:
> do you mean circuitpython with the rotary encoder qt board?

The rotary encoder qt board 4991.  I have it running with one and I soldered the back to get a second i2c # , but my limited tweeks did not work ☹

https://learn.adafruit.com/adafruit-i2c-qt-rotary-encoder/overview

ladyada, 2022-02-01:
do you mean circuitpython

troylball, 2022-02-01:
Yes I guess I did,    I am using the code from  https://learn.adafruit.com/adafruit-i2c-qt-rotary-encoder/overview wit a pi Zero 2 w and maybe a pi 4.

But that gest me wondering is there just regular python code as  well?   If so I would like to see that as well.

Cheers,
Troy