adafruit/seesaw #62: Negative Motor Throttle Values Not Working

aryker, 2022-12-21:
I am having a strange issue with my Crickit HAT. I have it connected to a Raspberry Pi 4 2GB, and I am using the Tri-Layer Mini Round Robot Chassis Kit to experiment. I have both DC motors hooked up the motor driver on the Crickit Hat. I am able to use the adafruit_crickit Python library to drive the motors forward without a problem. However, when I set the throttle values to a negative value, the motors do not move. I looked at the output from the Crickit HAT on an oscilloscope, and found that while positive values produce a typical PWM waveform, negative values do not produce any discernible signal. Below is the simple test I ran to confirm this result:

```
#!/usr/bin/env python3

from adafruit_crickit import crickit

MOTOR_R = crickit.dc_motor_1
MOTOR_L = crickit.dc_motor_2

MOTOR_L.throttle = 0.5 # Works correctly
MOTOR_R.throttle = -0.5 # Does not work!
```
I have updated my Crickit HAT firmware to the latest release. Am I missing something? Do I need to get a replacement?

caternuson, 2022-12-21:
Cross posted:
https://forums.adafruit.com/viewtopic.php?t=197159

aryker, 2022-12-22:
This was a wiring mistake on my part.

closed 2022-12-22 by aryker